FYNCH-BIO/evolver-electron #69: When app is opened, check electron memory for potential running expts / restart

zheins, 2020-12-30:


brandogw, 2021-01-11:
Decision tree for closing and opening the app.

[image]

zheins, 2021-01-11:
START
- [ ] Loading the experiment map from memory
- [ ] If things things exist in map - pop up the modal YES/NO
- [ ] if YES, loop through them, check if they actually exist, kill if they do AND are actually eVOLVER script, then restart
- [ ] If NO, loop through them and kill if actually eVOLVER script.

STOP

- [ ] Change existing pop-up to put the proper message (Don't use modal) Warning users of running expts (list)
- [ ] If yes,  continue to shut down, if no, return control.

zheins, 2021-01-14:
Clean up all of the random lists in the main process for paused/stopped/exptmap/running/etc.

closed 2021-01-25 by zheins