FYNCH-BIO/evolver-electron #97: Pump Cal GUI

zheins, 2021-01-19:


brandogw, 2021-02-01:
[https://app.zenhub.com/files/162771103/72b5dfb5-8a4c-4a3b-887d-88db36814afc/download](https://app.zenhub.com/files/162771103/72b5dfb5-8a4c-4a3b-887d-88db36814afc/download)

brandogw, 2021-02-01:
[image]

[image]

[image]

[https://app.zenhub.com/files/162771103/6177d27c-f9ed-4d74-86d4-15bc347becdc/download](https://app.zenhub.com/files/162771103/6177d27c-f9ed-4d74-86d4-15bc347becdc/download)

zheins, 2021-02-09:
PR #119 added core functionality of this issue. Last steps are just some styling.

closed 2021-02-22 by zheins